JuliaPhysics/SolidStateDetectors.jl #58: Configuration file splitting

ffischer42, 2020-10-08:
We should implement a way to link other JSON files in one JSON config file to outsource larger fractions and make it easy to switch parts.

**Status of the code:** _ready_

**To-Do:** 

- [ ] Function name decision: 

`scan_dict!()`, `deep_merge!()`, `recursive_merge!()` ,... ?

- [ ] Folder in the source code
- [ ] Test for the function. Should there be a test added?
- [ ] Implementation in SSD. Suggestion in function:

`parse_config_file(filename::AbstractString)::Dict{Any,Any}`

at return

`parsed_dict` -> `scan_dict!(parsed_dict)`

- [ ] Documentation has to be written
- [ ] Should this be an internal or exported function?

**I can do most of it**, but I need answers to the points above from you.

I attached an example configuration with  a jupyter notebook:

[json-merger.zip](https://github.com/JuliaPhysics/SolidStateDetectors.jl/files/5351277/json-merger.zip)

lmh91, 2020-10-30:
So in my opinion these are all internal functions and none of those should be exported. 
Also, I would extend the names of the functions so that their names actually describe what they are doing,
what makes it easier to understand the code. They are not supposed to be userfriendly (short) as they are not exported. 
E.g., `scan_dict!`, is quite generic. This could mean anything. What are you scanning for? 

I guess regarding testing we should add some example folder where one
detector, e.g. the IVC, is splitted in 3 or 4 configuration files. One holding the detector itself, 
one the world/grid (here maybe two config files where one is cartesian and one cylindrical), and one the cryostat.

ffischer42, 2020-11-11:
So after offline discussions the function name will be:
`scan_and_merge_included_json_files!(dict)`

I'll prepare the test config and add it to my upcoming pull request.

ffischer42, 2020-11-13:
Test configuration is created and a test was added to the runtests.jl

Test Summary:       | Pass  Total
Test real detectors |    4      4

lmh91, 2020-11-19:
This has been done with #104 
Thanks @ffischer42

closed 2020-11-19 by lmh91