JuliaPhysics/SolidStateDetectors.jl #61: Charge cloud effects: Diffusion & self-repulsion

lmh91, 2020-10-09:


lmh91, 2020-11-02:
#43 is related

fhagemann, 2020-11-13:
To implement charge cloud effects in SSD, I am thinking about implementing a new scalar fields for the electron and hole densities that evolve in time in `Event` rather than having point-like charges drifting along a drift path.

Find a short proposal on my approach on implementing it [here](https://github.com/JuliaPhysics/SolidStateDetectors.jl/files/5535709/Charge_cloud_effects_in_SSD.pdf).

fhagemann, 2020-11-24:
Diffusion and self-repulsion can also be implementing by extending our current implementation with point-like charges. Then, charge clusters would be simulated as some distribution of point-like charges with weighted charge values.

Find a second proposal for implementing diffusion and self-repulsion 
[here](https://github.com/JuliaPhysics/SolidStateDetectors.jl/files/5590240/Charge_cloud_effects_in_SSD_2.pdf).

fhagemann, 2021-01-12:
> Diffusion and self-repulsion can also be implemented by extending our current implementation with point-like charges. Then, charge clusters would be simulated as some distribution of point-like charges with weighted charge values.

I have something in the making. However, it is still not (yet) performant and needs some cross-check with analytic solutions to find the corresponding parameters for the diffusion.

Diffusion is implemented by a random walk with the same step length for each point charge. 
Self-repulsion is right now accounted for by calculated the electric field between pairs of charges and adding that to the external field. This requires to calculate the drift velocities for each time step rather than using the interpolated precalculated drift fields.

[image]

Using the platonic bodies twice around the charge cloud centre as initial point charge distributions results in some promising GIFs:

[image]
[image]
[image]
[image]
[image]

I will work on some speed-up that I have in mind and then push everything to the branch `refactor_drift_code` of my fork.

fhagemann, 2021-08-05:
The runtime of an N-body problem classically scales with O(N²). 
However, faster runtimes of O(N log N) can be achieved with a good choice of algorithms, e.g. [Barnes-Hut simulations](https://en.wikipedia.org/wiki/Barnes%E2%80%93Hut_simulation).

fhagemann, 2021-09-17:
What is still to be done for v0.7:
- [ ] Put the diffusion length to `MaterialProperties` (only for `HPGe` for now, `Si` can be added later)
- [ ] Handle charge clouds close initially close to the surface to avoid initial points being outside of the detector
- [ ] Write documentation for the manual such that people can use this experimental feature