JuliaPhysics/SolidStateDetectors.jl #68: Improve IO regarding flexibility

lmh91, 2020-10-09:


fraseyboo, 2021-03-12:
Regarding this issue it might be useful to support writing VTK files using VTKWrite.jl, I currently use VTK for a lot of my PSA work because the wide support for large data types is unparalleled. All the fields and electrodes should be relatively simple to export as rectilinear grids and polydata and can be easily interpreted externally in Paraview, Visit, Slicer or a native VTK program. 

[image]
[image]

oschulz, 2021-03-13:
Sounds very useful - we should add it as an optional dependency via Requires, though.

closed 2021-08-23 by fhagemann