JuliaPhysics/SolidStateDetectors.jl #75: Radial charge density gradients in cartesian space

leenderthayen, 2020-10-21:
Hi,

It seems like when specifying a radial charge distribution gradient in a cartesian grid, it takes the 'r' argument and turns it into an x gradient instead.

[image]

fhagemann, 2020-10-22:
Thanks for reporting, I will have a look at this today or tomorrow.

lmh91, 2020-11-02:
I have been thinking about this. 

We, actually should add something like a `RadialChargeDensityModel`.
Or, better something like a `StandardCrystalChargeDensityModel` for pulled crystals. 
Because the name `RadialChargeDensityModel` would conflict with a gradient in `z`. 

And LinearChargeDensityModel should only take `x`,`y`,`z`.

PR #88 (2020-11-06): Introduce new ChargeDensityModel that separates Cartesian and cylindrical gradients
This pull request is an alternative to #81 to fix #75.

A new struct `RadialChargeDensityModel` was added which is used when the impurity density changes linearly with `r` and/or `z`. Thus, it is the cylindrical equivalence to `LinearChargeDensityModel` which now only considers `x`, `y` and `z`.
Warnings are added, if `LinearChargeDensityModel` is parsed with `r` or if `RadialChargeDensityModel` is parsed with either `x` or `y`.

This might add a conversion of `CylindricalPoint` to `CartesianPoint` and vice ver [... cut]
closed 2020-11-06 by lmh91