JuliaPhysics/SolidStateDetectors.jl #79: Fix for hexagon primitive plot recipe

rjtayl, 2020-10-27:
Hi,

I noticed a mistake in the hexagon primitive that was added recently and I thought it would be easier to fix through here than a full pull request.

In HexagonalPrism.jl line 76 should be
`for φ in [deg2rad(30),deg2rad(90), deg2rad(150), deg2rad(210), deg2rad(270), deg2rad(330)]`
Currently the plot recipe is rotated 30 degrees relative to the rest of the geometries functions.

For example,
[image]
versus
[image]

oschulz, 2020-10-28:
Thanks @rjtayl . Could you actually make a PR, if possible? We always want the tests to run anyway.

lmh91, 2020-10-29:
This has been solved in #82.

closed 2020-10-29 by lmh91