JuliaPhysics/SolidStateDetectors.jl #84: CSG: Change internal Primitive Definition: Centered around 0

lmh91, 2020-10-30:
We should change our definitions of all primitives to be centered around (0,0,0).
Like in GEANT4. 

This also simplifies the rotation of objects (see #62). 

It also comes in handy when we want to add some clustering of objects onto subregions of a grid
in order to parallelize and speed up the initialization of the boundary conditions on the grid for the SOR
which can take some time if there are to many objects and the grid becomes fine. 
This clustering might also be used to speed up the charge drift in general (lesser lookups).

lmh91, 2020-10-30:
I added this issues to the v1.0.0 milestone as this might impact the definition of primitives in the config files. 
Although I am not sure about that.

fhagemann, 2020-12-03:
I was just thinking: the new CSG will probably only affect `Tube` and `Cone` when it comes to reading in config files.
However, the `z`-interval is defined with a field `"h"` right now. 

We could think about getting rid of `"h"` in general and only stick to `"z"` (either pass an interval with `"from"` and `"to"`, or just a number if centered around zero) and provide a conversion function that converts old CSG config files into new ones for the users to update.

lmh91, 2020-12-03:
I believe you can already do use "from" and "to". We had this at the very beginning. @oschulz wanted the `h`-syntax to be possible... :stuck_out_tongue_closed_eyes: 

We could add such a conversion function. User would probably appreciate it. There we could also introduce the change regarding the impurity-charge-density naming change.

oschulz, 2020-12-03:
`z` may be misleading especially when the object is centered at zero - is it the full height, or just from center to top. I think `h` is clearer.

fhagemann, 2021-01-12:
As we might need to change the config file layout anyhow, I worked on some functions to convert geometries to a dictionary that can be saved as config file afterwards. (Still experimental and trying to take into account a [comment from @lmh91](https://github.com/JuliaPhysics/SolidStateDetectors.jl/pull/122#issuecomment-739755931)).

```julia
using SolidStateDetectors
const SSD = SolidStateDetectors
const CSG = SSD.ConstructiveSolidGeometry
using Rotations
using StaticArrays
using IntervalSets
using DataStructures: OrderedDict

function construct_dict_from_geometry(g::CSG.AbstractGeometry{T}) where {T}
    d = OrderedDict{Any,Any}()
    add_entry_to_config_dict!(d, g)
    d
end

function add_entry_to_config_dict!(d::OrderedDict, g::CSG.TranslatedGeometry{T}) where {T}
    d["translate"] = OrderedDict{Any,Any}()
    t = g.t
    if t.x != 0 d["translate"]["x"] = t.x end
    if t.y != 0 d["translate"]["y"] = t.y end
    if t.z != 0 d["translate"]["z"] = t.z end
    add_entry_to_config_dict!(d["translate"], g.p)
    d
end

function add_entry_to_config_dict!(d::OrderedDict, g::CSG.ScaledGeometry{T}) where {T}
    d["scale"] = OrderedDict{Any,Any}()
    s = inv.(g.inv_s)
    if s[1] != 1 d["scale"]["x"] = s[1] end
    if s[2] != 1 d["scale"]["y"] = s[2] end
    if s[3] != 1 d["scale"]["z"] = s[3] end
    add_entry_to_config_dict!(d["scale"], g.p)
    d
end

function add_entry_to_config_dict!(d::OrderedDict, g::CSG.RotatedGeometry{T}) where {T}
    d["rotate"] = OrderedDict{Any,Any}()
    r = inv.(g.inv_r)
    
    # To Do: store rotation
    
    add_entry_to_config_dict!(d["rotate"], g.p)
    d
end

function add_entry_to_config_dict!(d::OrderedDict, g::CSG.CSGUnion{T}) where {T}
    d["union"] = OrderedDict[]
    add_entry_to_union_config_dict!(d["union"], g.a)
    add_entry_to_union_config_dict!(d["union"], g.b)
    d
end

function add_entry_to_union_config_dict!(d::Vector{OrderedDict}, g::CSG.AbstractGeometry{T}) where {T}
    dict = OrderedDict()
    add_entry_to_config_dict!(dict, g)
    push!(d, dict)
    d
end

function add_entry_to_union_config_dict!(d::Vector{OrderedDict}, g::CSG.CSGUnion{T}) where {T}
    add_entry_to_union_config_dict!(d, g.a)
    add_entry_to_union_config_dict!(d, g.b)
end
    
        
function add_entry_to_config_dict!(d::OrderedDict, g::CSG.CSGDifference{T}) where {T}
    d["difference"] = OrderedDict[]
    pos = OrderedDict()
    add_entry_to_config_dict!(pos, g.a)
    push!(d["difference"], pos)
    neg = OrderedDict()https://github.com/JuliaPhysics/SolidStateDetectors.jl/pull/122#issuecomment-739755931
    add_entry_to_config_dict!(neg, g.b)
    push!(d["difference"], neg)
    d
end

function add_entry_to_config_dict!(d::OrderedDict, g::CSG.Cone{T,<:Tuple,<:Any,<:Any}) where {T}
    d["cone"] = OrderedDict()
    d["cone"]["r"] = OrderedDict()
    d["cone"]["r"]["bottom"] = typeof(g.r[1]) == T ? g.r[1] : OrderedDict("from" => g.r[1].left, "to" => g.r[1].right)
    d["cone"]["r"]["top"] = typeof(g.r[2]) == T ? g.r[2] : OrderedDict("from" => g.r[2].left, "to" => g.r[2].right)
    if !isnothing(g.φ) d["cone"]["phi"] = OrderedDict("from" => g.φ.left, "to" => g.φ.right) end
    typeof(g.z) == T ? d["cone"]["h"] = g.z * 2 : dict["cone"]["z"] = OrderedDict("from" => g.z.left, "to" => g.z.right) 
    d
end
    
function add_entry_to_config_dict!(d::OrderedDict, g::CSG.Cone{T,<:Union{T, AbstractInterval},<:Any,<:Any}) where {T}
    d["tube"] = OrderedDict()
    d["tube"]["r"] = typeof(g.r) == T ? g.r : OrderedDict("from" => g.r.left, "to" => g.r.right)
    if !isnothing(g.φ) d["tube"]["phi"] = OrderedDict("from" => g.φ.left, "to" => g.φ.right) end
    typeof(g.z) == T ? d["tube"]["h"] = g.z * 2 : dict["tube"]["z"] = OrderedDict("from" => g.z.left, "to" => g.z.right) 
    d
end
```

This would allow to create geometries in the REPL/script and save them afterwards to a config file.

Example:
```julia
using YAML
test = ((CSG.RotMatrix(RotX(π/4)) * CSG.Tube(1,1)) - CSG.scale(CSG.Tube(0.5,1), SVector{3,Float64}(1,1,0.5))) + CSG.CartesianVector{Float64}(0,0,-1)
d = construct_dict_from_geometry(test)
YAML.write_file("test_config_file.yaml", d)
```

results in:
```yaml
translate:
  z: -1.0
  difference:
    - rotate:
        tube:
          r: 1.0
          h: 1.0
    - scale:
        z: 0.5
        tube:
          r: 0.5
          h: 1.0
```

However, we need to come up with a consistent way to define rotations in the config file (#62).

lmh91, 2021-01-12:
Regarding the rotation:
I talked with @hervasa2 about this some weeks ago. Though I do not remember the details.
I believe we had some idea like:
```yaml
- rotate: XYZ = [90, 0, 45] # first rotate around X, than, Y, than Z
- rotate: ZX = [-45, 10] # first rotate around Z, than X
```
it should also be possible to specify a Rotation matrix
```yaml
- rotate: M = [1, 0, 0, 0, 1, 0, 0, 0, 0] # Vector of length 9 (don't know if matrix would is supported by yaml/json)
```
Of course, the second field would be the object which should be rotated. 

This syntax would require to analyse the keys of the rotation dictionary though.

closed 2021-04-20 by fhagemann