JuliaPhysics/SolidStateDetectors.jl #87: Clean up: remove obsolte functions

fhagemann, 2020-11-02:
Some of the functions are still left over from refactoring the JSON files.

One example is the [`bounding_box`](https://github.com/JuliaPhysics/SolidStateDetectors.jl/blob/master/src/SolidStateDetector/DetectorGeometries.jl#L420) function, which returns the `crystal_radius` and `crystal_length` which is no longer stored in `SolidStateDetector`. 

Also, we should look for functions that are defined but never called, e.g. [`from_internal_units`](https://github.com/JuliaPhysics/SolidStateDetectors.jl/blob/master/src/Geometries/Units.jl#L21).

These functions should be either removed or incorporated accordingly.

lmh91, 2020-11-02:
Yes, `from_internal_units` is not really used or just at some points. 
At the beginning we did not used Unitful.jl that much. 
But we should for external functions. Internally we should convert all the units to SI units using `to_internal_units`
and afterwards return the output in the units corresponding to the given input units via `from_internal_units`.

This addresses for example the simulation of events, see #66.

fhagemann, 2020-11-12:
How important are the functions `get_important_points` for the volume primitives? 
Apparently, they were used to adapt the grid to the volumes before implementing the painting algorithm.
Are they still required?

lmh91, 2020-11-12:
They are extremely important.

fhagemann, 2020-11-28:
I came across two functions that are not used and, thus, obsolete (please confirm!)
- `is_real` in [ChargeDrift.jl, line 173](https://github.com/JuliaPhysics/SolidStateDetectors.jl/blob/master/src/ChargeDrift/ChargeDrift.jl#L173) is not called anywhere
- `SolidStateDetector{T}` in [SolidStateDetector.jl, line 245](https://github.com/JuliaPhysics/SolidStateDetectors.jl/blob/master/src/SolidStateDetector/SolidStateDetector.jl#L245) seems to call itself (?)

fhagemann, 2021-01-13:
The function ` _drift_charge(...)` in [ChargeDrift.jl line 60](https://github.com/JuliaPhysics/SolidStateDetectors.jl/blob/master/src/ChargeDrift/ChargeDrift.jl#L60) seems to be an internal function but is never called. Is it obsolete?

schustermartin, 2021-01-13:
All it does is to put the starting position in a vector and call the _drift_charges(...) function. Basically, the application would be to call the function for a single starting position, warranting the singular, which can be a legit argument for user friendliness. This is an internal function though, so as far as I am concerned it can go.

fhagemann, 2021-07-21:
I have created a new `cleanup` branch that we should use to remove obsolete functions and add docstrings to all (exported) functions that remain.

lmh91, 2021-07-28:
I think this can be actually closed, right?

fhagemann, 2021-07-28:
Yes, I will close this.

closed 2021-07-28 by fhagemann