JuliaPhysics/SolidStateDetectors.jl #92: Geometry plotting

leenderthayen, 2020-11-12:
Hello everyone,

I was wondering whether it's possible to add the plotting of the full geometry rather than just the contacts. We want to implement a somewhat more involved geometry and debugging correct positions is quite tricky without visualization or boundary checking methods. I assume defining everything as a contact for debugging would be considered bad practice, but something like that would be of great help.

Thanks!

lmh91, 2020-11-12:
This is possible. Though, not in one line. 
We went for only plotting the positive primitives of all contacts for reasons of clarity. 

All Primitives are stored in your `SolidStateDetector`-object (`sim.detector`).
The `SolidStateDetector` struct has the fields `semiconductors`, `contacts` and `passives` (and `virtual_drift_volumes`).
These are vectors of objects of type `Semiconductor`, `Contact` and `Passive`. These all have fields `geometry`. But also two fields, `geometry_positive` and `geometry_negative`, which are vectors of primitive which you can plot via our recipes.

fhagemann, 2021-07-29:
In our new release (v0.6.0), we also added plot recipes for all objects. However, `plot(sim.detector)` will still only plot contacts and **not** the semiconductor.

For our example detector with configuration file at `SSD_examples[:CGD]`:

```julia
plot(sim.detector)
```
[image]

```julia
plot(sim.detector)
plot!(sim.detector.semiconductor)
```

[image]

Note that `geometry_positive` and `geometry_negative` do not longer exist in the new release v0.6.0.
Feel free to have a look through our updated documentation and report all kinds of bugs that you might find! ;)

closed 2021-08-04 by fhagemann