JuliaPhysics/SolidStateDetectors.jl #93: Electric field plotting in Cartesian geometry

leenderthayen, 2020-11-12:
Hello everyone,

When running the plot_electric_field(!) functions on a simulation with a cartesian geometry it does not show the expected behaviour. Instead, it outputs this
[ Info: 0 drifts are now being simulated...
┌ Warning: Attribute alias `ylabel` detected in the user recipe defined for the signature (::SolidStateDetectors.Plot_electric_field). To ensure expected behavior it is recommended to use the default attribute `yguide`.
└ @ Plots ~/.julia/packages/Plots/uCh2y/src/pipeline.jl:15
┌ Warning: Attribute alias `xlabel` detected in the user recipe defined for the signature (::SolidStateDetectors.Plot_electric_field). To ensure expected behavior it is recommended to use the default attribute `xguide`.
└ @ Plots ~/.julia/packages/Plots/uCh2y/src/pipeline.jl:15

And simply shows the electric potential with no field lines. The electric field calculation itself is fine to my knowledge.

Thanks!

schustermartin, 2020-11-12:
Hi Leendert! Yes, the plot recipies for the electric field are not really working as intended at the moment, also since there have been a couple of changes and updates to the Plots.jl package. I am currently working on this and hopefully will have them back working very soon.
Cheers Martin

leenderthayen, 2020-11-12:
Great, thanks for the swift reply!

schustermartin, 2020-11-19:
Dear Leendert, 
did pull request #101 resolve your issue?

leenderthayen, 2020-11-20:
Dear Martin,

Very nice update! My only comment would be that plotting fieldlines breaks when you don't specify the sampling, at least when I tested it using the hexagon example. 

`ERROR: MethodError: no method matching searchsortednearest(::Array{Any,1}, ::Float32)
Closest candidates are:
  searchsortednearest(::SolidStateDetectors.DiscreteAxis{T,:periodic,:periodic,I} where I, ::T) where T<:Real at /Users/leenderthayen/.julia/packages/SolidStateDetectors/Fyds7/src/Axes/DiscreteAxis.jl:331
  searchsortednearest(::SolidStateDetectors.DiscreteAxis{T,BL,BR,I} where I where BR where BL, ::T) where T<:Real at /Users/leenderthayen/.julia/packages/SolidStateDetectors/Fyds7/src/Axes/DiscreteAxis.jl:328
  searchsortednearest(::AbstractArray{T,1}, ::T) where T<:Real at /Users/leenderthayen/.julia/packages/SolidStateDetectors/Fyds7/src/Axes/DiscreteAxis.jl:317`

I assume this can be fixed easily by using some default values.

Cheers,
Leendert

schustermartin, 2020-11-20:
Dear Leendert, thank you for your comment. Yes, this error arises because the default value for sampling is chosen so big that the first sampling step leaves the contact. Then the array storing the sampled points is not well populated (if at all) and the `searchsortednearest` method fails (as it should in this case). In your case the sampled points array really is empty as the passed array has type `Any`.
I have implemented a fix, i.e. automatic down scaling of the step size if this happens. I will create a pull request shortly. Thanks again for the input.

PR #108 (2020-11-23): Electric field plot enhancement
This pull request will close #93.

- Adding a automatic down scaling of the `sampling` step if it is too big and would lead to unpopulated arrays. The `offset` is scaled accordingly
- Revising default values if no cross-section is passed

Now, for the Hexagon Example detector, the lines
```
plot(sim.electric_field)
plot_electric_fieldlines!(sim)
```
yield
[image]
and a corresponding warning that the specified/default values for `sampling` and `offset` have been modified automatically.
closed 2020-11-23 by schustermartin