JuliaPhysics/SolidStateDetectors.jl #94: HDF5 IO integration

leenderthayen, 2020-11-12:
Hello everyone,

It appears that currently the IO capabilities for HDF5 saving and loading are located in an unregistered package, LegendHDF5IO. Is it possible to integrate this directly into SSD, or if not bind the github repo's together?

Thanks!

lmh91, 2020-11-12:
So what SSD provides is a conversion (and reconversion) of several internal structs into NamedTuples.
LegendHDF5IO provides a generic IO of NamedTuples to HDF5. 
You are right, this should be an LEGEND independent official registered package. 
So far, we (@oschulz) had no time for this.

oschulz, 2020-11-12:
We have indeed been considering to turn (part of) LegendHDF5IO into a generic registered Julia package, but that will require some careful "surgery" and will probably still take a while. However, we wouldn't want to add heavy additional dependencies like HDF5 to SolidStateDetectors, so SolidStateDetectors still wouldn't depend on it. The current mechanism of conversion from/to NamedTuples gives the user freedom regarding which kind of I/O they want to use.

Having said that, I'm planning a custom package registry for LEGEND (since the LEGEND packages are too experiment-specific for the General registry). That will make it easier to install LegendHDF5IO in addition to SolidStateDetectors.

We should also check if JLD2 can be used out of the box (or with some tweaks) to store/load SolidStateDetectors detector definitions and fields. This may be an attractive option for non-LEGEND-users. There have been some improvements in JLD2 recently.

leenderthayen, 2021-03-24:
We are currently using JLD2, so we can close this for now.

closed 2021-03-24 by leenderthayen