JuliaPhysics/SolidStateDetectors.jl #107: Refactor Tests

lmh91, 2020-11-19:
At the moment, we are simulating quite a number of detectors in our tests. 
We should reduce that.

I guess besides the comparison to analytic solutions we only need to simulate 2 real detectors.
One for cartesian and one for cylindrical coordinates. 

We should add then smaller, more specific tests for the geometry, e.g., primitives. 
But we don't need a full simulation for each primitive.

schustermartin, 2020-11-19:
I second this.

I think there still should be a test script that basically runs every single feature and function we have to offer. This should be made available in the package but not as  the one that is  run automatically. People can run it locally for somewhat larger PRs to ensure nothing really breaks.

fhagemann, 2020-12-08:
This is already in the making! 
#122 already has script that tests primitives, transformations, sets and read-in.
Whenever we merge the new CSG with the already existing simulation code, the current and new test scripts can be combined.