JuliaPhysics/SolidStateDetectors.jl #119: PackageCompiler crashes

oschulz, 2020-12-04:
PackageCompiler now crashes when trying to compile a Julia system image that contains SolidStateDetectors and a simple precompile execution file like

```julia
using SolidStateDetectors

detector_config_filename = SSD_examples[:InvertedCoax]

# PackageCompiler crashes here:
simulation = Simulation{Float32}(detector_config_filename)
```

This used to work fine - not sure if this is due to changes in SolidStateDetectors, in one of it's depencencies or Julia v1.4 -> v1.5.

oschulz, 2020-12-04:
Here's a Gist to reproduce the problem: https://gist.github.com/oschulz/2f2daab0f9ce56d3086d55ad6fa18c19

Run with `JULIA_LOAD_PATH="@:@stdlib" julia --project=. build_sysimage.jl`.

Tested with Julia v1.5.3 on Linux-x86_84.

lmh91, 2020-12-05:
It crashes here:
```julia
    inputunits = dunits::Dict{String, Unitful.Units} = Dict{String, Unitful.Units}(
        "length" => u"m", # change this to u"m" ? SI Units
        "potential" => u"V",
        "angle" => u"°",
        "temperature" => u"K"
    )
```
I don't know why the first line looks like that...
However, changing it to 
```julia
```julia
    inputunits = Dict{String, Unitful.Units}(
        "length" => u"m", # change this to u"m" ? SI Units
        "potential" => u"V",
        "angle" => u"°",
        "temperature" => u"K"
    )
```
does not solve the problem...

lmh91, 2020-12-05:
New MWE:

Add `Uniful.jl` to the project and change `precompile_exec.jl` to 

```julia
using Unitful 
Dict{String, Unitful.Units}(
    "length" => u"m", # change this to u"m" ? SI Units
    "potential" => u"V",
    "angle" => u"°",
    "temperature" => u"K"
)
```

lmh91, 2020-12-05:
PackageCompiler does not like unit strings...

oschulz, 2020-12-05:
Thanks, that narrows it down nicely! Just reported it: JuliaLang/julia#38719

lmh91, 2020-12-05:
So, it seems we have to wait for Julia v1.5.4 or v1.6. 
I will close this.

closed 2020-12-05 by lmh91