JuliaPhysics/SolidStateDetectors.jl #131: Capacitance calculation and depletion

leenderthayen, 2021-01-08:
Hello,

We are looking at the C-V behaviour of our geometries, but it is not entirely clear how to combine the information from point types and the capacitance calculation. Specifically, the capacitance continues decreasing even though (visually) all point types show up as green. As an example, taking a 2mm long tube with contacts on either end and varying the applied voltage and setting the impurity concentration to , you obtain the following C-V and d(1/C^2)/dV curves

[image]

From about 800V onwards, however, the point types show up as green, even though the slope of the C-V is non-zero at 1000V.

At 120V
[image]

At 780V
[image]

All electric potentials are calculated with depletion_handling=true on. In the naive analytical result one would expect the detector to be fully depleted around 600V, though I can understand edge effects playing a role here.

Thanks!
Leendert

[big_tube.txt](https://github.com/JuliaPhysics/SolidStateDetectors.jl/files/5789678/big_tube.txt)

lmh91, 2021-01-11:
No, that is normal and also correct. 
The capacitance is given through (2x) the energy stored in the electric field divided through the squared bias voltage. 
Even, when the detector is fully depleted, the electric field still changes with increasing bias voltage. 

We have comparison to analytical solutions which will also be in a paper about SSD on which we are currently working.

leenderthayen, 2021-01-11:
This makes sense, thank you. 

For the record, am I correct in assuming that what the package does is calculate the three coupled differential equations, i.e. time-independent drift equations for p and n, and the poisson equation? The documentation only mentions the latter. The numbers in the charge density model correspond to the donor and acceptor atoms which are assumed to be fully ionized, is that correct?

leenderthayen, 2021-01-11:
The reason I am concerned wrt the previous question is that the new update seems to have brought qualitative changes with it when calculating the electric potential. Taking again a simple reverse biased tube with a near-intrinsic charge density and calculating the electric potential

[image]

[big_tube.txt](https://github.com/JuliaPhysics/SolidStateDetectors.jl/files/5798415/big_tube.txt)

I would expect the undepleted regions to have a constant electric potential equal to the potential at the ohmic contact, because while the donor and acceptor atoms can be assumed ionized, the crystal should still be electrically neutral in those parts. I don't understand what's going on here unless you only solve for the poisson equation.

fhagemann, 2021-01-11:
We do not use the time-independent drift equations for p and n, but only the Poisson equation.
When `depletion_handling = true`, then the charge density resulting from ionised impurities in the undepleted regions is set to zero, according to the case where the crystal is electrically neutral. 

I can only reproduce your plot when having `depletion_handling = false`, which is exactly what you assume that we "only solve for the poisson equation and take the fully ionized impurity density as the charge distribution obeying boundary conditions at contacts."  

My plot (with only one refinement):

[image]

Did you try to simulate the potential with `depletion_handling = true`? That should take into account that some impurities are not ionised and result in a constant electric potential in the undepleted region.

lmh91, 2021-01-12:
Some general note regarding capacity calculations. 
The capacity really needs a fine enough grid. 
In some of my simulation I use the capacity kinda as a refinement parameter. Thus, I refine until the capacity does not change much anymore. Then, I know my grid is fine enough. For now, this has do be done manually.

leenderthayen, 2021-01-12:
Thanks Felix and Lukas, that was very useful. The grid requirements for capacity calculations is a useful comment, and something I've noticed too. We normally do all of our calculations with depletion_handling=true by default, but I realized I didn't actually know what it was doing when enabling it. Final question before I close the thread: How do you determine whether a voxel is depleted?

lmh91, 2021-01-12:
Inside the semiconductor there cannot be a local extrema in the potential. The potential has to drop/rise along the fieldlines from one contact to the other monotonously. Each grid point has 6 direct neighbors. If the new potential value for this point, first calculated with the contribution of the charge density, is larger or smaller than the maximum/minimum of the potential values of the neighbors, the contribution of the charge density is substracted (set to 0).

closed 2021-01-12 by leenderthayen