JuliaPhysics/SolidStateDetectors.jl #132: Asymmetric artefacts for symmetric objects in cartesian grid

leenderthayen, 2021-01-08:
Hello,

When preparing toy examples for some of the previous issues posted today, I noticed that an asymmetry in the point types (and upon later inspection also on the electric potentials and every other quantity) was not a result of an asymmetric x-y geometry (like a hexagon), but also appears in the case of round tubes. As an example, I've attached the point types and electric potential of a simple tube in different planes.

[image]

[image]

[big_tube.txt](https://github.com/JuliaPhysics/SolidStateDetectors.jl/files/5789697/big_tube.txt)

Looking also at edges which are not parallel to x or y, there appear grid artefacts which can be tamed somewhat with increasing grid spacing, but remain visible.

In one case, this became particularly apparent

[image]

This is the central pixel of a hexagonal pixel ring. Every pixel contains roughly the same asymmetry, with a minimum along the x-axis through the center of the pixel and a butterfly structure around it.

Thank you!
Leendert

lmh91, 2021-01-11:
There was indeed a bug. Please try it again with the current master branch.

leenderthayen, 2021-01-11:
This seems to have fixed things nicely. Thank you for the fast response!

leenderthayen, 2021-01-11:
While it looks like things are fixed for spherically symmetric geometries, using for example a hexagonal pixel introducing further asymmetric grid artefacts.

[image]
[image]

[simple_si_pixel.txt](https://github.com/JuliaPhysics/SolidStateDetectors.jl/files/5798322/simple_si_pixel.txt)

lmh91, 2021-01-12:
Okay, I think there is no problem here. 
You have set `depletion_handling = true`, right?

These are just artefacts due to the depletion handling, which appear that strong on a coarse grid.
In the relaxation, the charge density in a voxel (around a grid point) can switched between its density value (depleted) to 0 (undepleted) at the border of the undepleted-to-depleted volumes or undepleted-to-outside-of-detector volumes.
This leads to a "big" change of the new potential value of that grid point in each iteration step of the relaxation. 
This can lead to such artefacts. This effect however, should become smaller if you refine your grid further. 

If I set the charge density to 0 in the configuration file, these artifacts also do not appear even with depletion_handling turned on 
as there is no "change" anymore in the density at any grid point.

leenderthayen, 2021-01-12:
That's very useful to know. Is there a good rule of thumb as to what grid sizes you should be aiming at without having to check visually? In #131 you mention using the capacitance stability, but if you have a lower boundary in mind that would also help.

lmh91, 2021-01-12:
Hmm, no not really. These are also things we are investigating and trying to improve. To find some better defaults (or algorithms, e.g. regarding how the grid is refined), which always lead to a good result. But the term "good" depends on to what level one wants to simulate something. 
Also, to some extend, performance has to be taken into account (at least regarding the default values).

leenderthayen, 2021-01-12:
I understand. In the absence of better algorithms or default values a way of quantifying artefacts might also be useful, such as sudden discrete steps. I'll close this for now.

closed 2021-01-12 by leenderthayen