JuliaPhysics/SolidStateDetectors.jl #134: Qt segmentation fault when plotting

leenderthayen, 2021-01-11:
Hello,

This might be more difficult to track down, but I've been experiencing segmentation faults over the past weeks when trying to plot quantities from SSD.

I have attached the full stack trace of the one that just occurred in the hope that you will be able to extract something useful.

`julia> plot(s.electric_potential, x=0)
QWidget::paintEngine: Should no longer be called
QPainter::begin: Paint device returned engine == 0, type: 1
QWidget::paintEngine: Should no longer be called
QPainter::begin: Paint device returned engine == 0, type: 1

signal (11): Segmentation fault: 11
in expression starting at REPL[21]:0
_ZN13QMacCGContextC2EP8QPainter at /Users/leenderthayen/.julia/conda/3/lib/libQt5Gui.5.9.7.dylib (unknown line)
_ZN23QPixmapDropShadowFilter11qt_metacallEN11QMetaObject4CallEiPPv at /Users/leenderthayen/.julia/conda/3/lib/libQt5Widgets.5.9.7.dylib (unknown line)
_ZN8QToolBar10paintEventEP11QPaintEvent at /Users/leenderthayen/.julia/conda/3/lib/libQt5Widgets.5.9.7.dylib (unknown line)
_ZN11sipQToolBar10paintEventEP11QPaintEvent at /Users/leenderthayen/.julia/conda/3/lib/python3.7/site-packages/PyQt5/QtWidgets.so (unknown line)
_ZN7QWidget5eventEP6QEvent at /Users/leenderthayen/.julia/conda/3/lib/libQt5Widgets.5.9.7.dylib (unknown line)
_ZN8QToolBar5eventEP6QEvent at /Users/leenderthayen/.julia/conda/3/lib/libQt5Widgets.5.9.7.dylib (unknown line)
_ZN11sipQToolBar5eventEP6QEvent at /Users/leenderthayen/.julia/conda/3/lib/python3.7/site-packages/PyQt5/QtWidgets.so (unknown line)
_ZN19QApplicationPrivate13notify_helperEP7QObjectP6QEvent at /Users/leenderthayen/.julia/conda/3/lib/libQt5Widgets.5.9.7.dylib (unknown line)
_ZN12QApplication6notifyEP7QObjectP6QEvent at /Users/leenderthayen/.julia/conda/3/lib/libQt5Widgets.5.9.7.dylib (unknown line)
_ZN15sipQApplication6notifyEP7QObjectP6QEvent at /Users/leenderthayen/.julia/conda/3/lib/python3.7/site-packages/PyQt5/QtWidgets.so (unknown line)
_ZN16QCoreApplication15notifyInternal2EP7QObjectP6QEvent at /Users/leenderthayen/.julia/conda/3/lib/libQt5Core.5.9.7.dylib (unknown line)
_ZN14QWidgetPrivate14sendPaintEventERK7QRegion at /Users/leenderthayen/.julia/conda/3/lib/libQt5Widgets.5.9.7.dylib (unknown line)
_ZN14QWidgetPrivate10drawWidgetEP12QPaintDeviceRK7QRegionRK6QPointiP8QPainterP19QWidgetBackingStore at /Users/leenderthayen/.julia/conda/3/lib/libQt5Widgets.5.9.7.dylib (unknown line)
_ZN14QWidgetPrivate22paintSiblingsRecursiveEP12QPaintDeviceRK5QListIP7QObjectEiRK7QRegionRK6QPointiP8QPainterP19QWidgetBackingStore at /Users/leenderthayen/.julia/conda/3/lib/libQt5Widgets.5.9.7.dylib (unknown line)
_ZN14QWidgetPrivate22paintSiblingsRecursiveEP12QPaintDeviceRK5QListIP7QObjectEiRK7QRegionRK6QPointiP8QPainterP19QWidgetBackingStore at /Users/leenderthayen/.julia/conda/3/lib/libQt5Widgets.5.9.7.dylib (unknown line)
_ZN14QWidgetPrivate22paintSiblingsRecursiveEP12QPaintDeviceRK5QListIP7QObjectEiRK7QRegionRK6QPointiP8QPainterP19QWidgetBackingStore at /Users/leenderthayen/.julia/conda/3/lib/libQt5Widgets.5.9.7.dylib (unknown line)
_ZN14QWidgetPrivate10drawWidgetEP12QPaintDeviceRK7QRegionRK6QPointiP8QPainterP19QWidgetBackingStore at /Users/leenderthayen/.julia/conda/3/lib/libQt5Widgets.5.9.7.dylib (unknown line)
_ZN14QWidgetPrivate10scrollRectERK5QRectii at /Users/leenderthayen/.julia/conda/3/lib/libQt5Widgets.5.9.7.dylib (unknown line)
_ZN14QWidgetPrivate10scrollRectERK5QRectii at /Users/leenderthayen/.julia/conda/3/lib/libQt5Widgets.5.9.7.dylib (unknown line)
_ZN14QDesktopWidget11qt_metacallEN11QMetaObject4CallEiPPv at /Users/leenderthayen/.julia/conda/3/lib/libQt5Widgets.5.9.7.dylib (unknown line)
_ZN14QDesktopWidget11qt_metacallEN11QMetaObject4CallEiPPv at /Users/leenderthayen/.julia/conda/3/lib/libQt5Widgets.5.9.7.dylib (unknown line)
_ZN19QApplicationPrivate13notify_helperEP7QObjectP6QEvent at /Users/leenderthayen/.julia/conda/3/lib/libQt5Widgets.5.9.7.dylib (unknown line)
_ZN12QApplication6notifyEP7QObjectP6QEvent at /Users/leenderthayen/.julia/conda/3/lib/libQt5Widgets.5.9.7.dylib (unknown line)
_ZN15sipQApplication6notifyEP7QObjectP6QEvent at /Users/leenderthayen/.julia/conda/3/lib/python3.7/site-packages/PyQt5/QtWidgets.so (unknown line)
_ZN16QCoreApplication15notifyInternal2EP7QObjectP6QEvent at /Users/leenderthayen/.julia/conda/3/lib/libQt5Core.5.9.7.dylib (unknown line)
_ZN22QGuiApplicationPrivate18processExposeEventEPN29QWindowSystemInterfacePrivate11ExposeEventE at /Users/leenderthayen/.julia/conda/3/lib/libQt5Gui.5.9.7.dylib (unknown line)
_ZN22QWindowSystemInterface22sendWindowSystemEventsE6QFlagsIN10QEventLoop17ProcessEventsFlagEE at /Users/leenderthayen/.julia/conda/3/lib/libQt5Gui.5.9.7.dylib (unknown line)
_ZN22QWindowSystemInterface23flushWindowSystemEventsE6QFlagsIN10QEventLoop17ProcessEventsFlagEE at /Users/leenderthayen/.julia/conda/3/lib/libQt5Gui.5.9.7.dylib (unknown line)
qt_plugin_instance at /Users/leenderthayen/.julia/conda/3/plugins/platforms/libqcocoa.dylib (unknown line)
_ZN7QWindow10setVisibleEb at /Users/leenderthayen/.julia/conda/3/lib/libQt5Gui.5.9.7.dylib (unknown line)
_ZN14QWidgetPrivate8show_sysEv at /Users/leenderthayen/.julia/conda/3/lib/libQt5Widgets.5.9.7.dylib (unknown line)
_ZN14QWidgetPrivate11show_helperEv at /Users/leenderthayen/.julia/conda/3/lib/libQt5Widgets.5.9.7.dylib (unknown line)
_ZN7QWidget10setVisibleEb at /Users/leenderthayen/.julia/conda/3/lib/libQt5Widgets.5.9.7.dylib (unknown line)
_ZN14sipQMainWindow10setVisibleEb at /Users/leenderthayen/.julia/conda/3/lib/python3.7/site-packages/PyQt5/QtWidgets.so (unknown line)
_ZL17meth_QWidget_showP7_objectS0_ at /Users/leenderthayen/.julia/conda/3/lib/python3.7/site-packages/PyQt5/QtWidgets.so (unknown line)
_PyMethodDef_RawFastCallKeywords at /Users/leenderthayen/.julia/conda/3/lib/libpython3.7m.dylib (unknown line)
_PyCFunction_FastCallKeywords at /Users/leenderthayen/.julia/conda/3/lib/libpython3.7m.dylib (unknown line)
call_function at /Users/leenderthayen/.julia/conda/3/lib/libpython3.7m.dylib (unknown line)
_PyEval_EvalFrameDefault at /Users/leenderthayen/.julia/conda/3/lib/libpython3.7m.dylib (unknown line)
function_code_fastcall at /Users/leenderthayen/.julia/conda/3/lib/libpython3.7m.dylib (unknown line)
call_function at /Users/leenderthayen/.julia/conda/3/lib/libpython3.7m.dylib (unknown line)
_PyEval_EvalFrameDefault at /Users/leenderthayen/.julia/conda/3/lib/libpython3.7m.dylib (unknown line)
_PyEval_EvalCodeWithName at /Users/leenderthayen/.julia/conda/3/lib/libpython3.7m.dylib (unknown line)
_PyFunction_FastCallDict at /Users/leenderthayen/.julia/conda/3/lib/libpython3.7m.dylib (unknown line)
_PyObject_Call_Prepend at /Users/leenderthayen/.julia/conda/3/lib/libpython3.7m.dylib (unknown line)
method_call at /Users/leenderthayen/.julia/conda/3/lib/libpython3.7m.dylib (unknown line)
PyObject_Call at /Users/leenderthayen/.julia/conda/3/lib/libpython3.7m.dylib (unknown line)
macro expansion at /Users/leenderthayen/.julia/packages/PyCall/tqyST/src/exception.jl:95 [inlined]
#109 at /Users/leenderthayen/.julia/packages/PyCall/tqyST/src/pyfncall.jl:43 [inlined]
disable_sigint at ./c.jl:446 [inlined]
__pycall! at /Users/leenderthayen/.julia/packages/PyCall/tqyST/src/pyfncall.jl:42 [inlined]
_pycall! at /Users/leenderthayen/.julia/packages/PyCall/tqyST/src/pyfncall.jl:29
_pycall! at /Users/leenderthayen/.julia/packages/PyCall/tqyST/src/pyfncall.jl:11 [inlined]
#_#116 at /Users/leenderthayen/.julia/packages/PyCall/tqyST/src/pyfncall.jl:86 [inlined]
PyObject at /Users/leenderthayen/.julia/packages/PyCall/tqyST/src/pyfncall.jl:86
_display at /Users/leenderthayen/.julia/packages/Plots/uCh2y/src/backends/pyplot.jl:1376
[... 13 lines cut ...]
jl_apply at /Users/julia/buildbot/worker/package_macos64/build/src/./julia.h:1700 [inlined]
do_apply at /Users/julia/buildbot/worker/package_macos64/build/src/builtins.c:643
jl_f__apply at /Users/julia/buildbot/worker/package_macos64/build/src/builtins.c:657 [inlined]
jl_f__apply_latest at /Users/julia/buildbot/worker/package_macos64/build/src/builtins.c:693
#invokelatest#1 at ./essentials.jl:712 [inlined]
invokelatest at ./essentials.jl:711 [inlined]
run_interface at /Users/julia/buildbot/worker/package_macos64/build/usr/share/julia/stdlib/v1.4/REPL/src/LineEdit.jl:2354
run_frontend at /Users/julia/buildbot/worker/package_macos64/build/usr/share/julia/stdlib/v1.4/REPL/src/REPL.jl:1055
run_repl at /Users/julia/buildbot/worker/package_macos64/build/usr/share/julia/stdlib/v1.4/REPL/src/REPL.jl:206
#764 at ./client.jl:383
[... cut ...]
jl_f__apply_latest at /Users/julia/buildbot/worker/package_macos64/build/src/builtins.c:693
#invokelatest#1 at ./essentials.jl:712 [inlined]
invokelatest at ./essentials.jl:711 [inlined]
run_main_repl at ./client.jl:367
exec_options at ./client.jl:305
_start at ./client.jl:484
jfptr__start_2075.clone_1 at /Applications/Julia-1.4.app/Contents/Resources/julia/lib/julia/sys.dylib (unknown line)
true_main at /usr/local/bin/julia (unknown line)
main at /usr/local/bin/julia (unknown line)
Allocations: 17013236864 (Pool: 17013122412; Big: 114452); GC: 4127
Segmentation fault: 11`

oschulz, 2021-01-12:
Could you try with [Julia vv1.6.0-beta1](https://julialang.org/downloads/#upcoming_release)? I think v1.6 includes some segfault-related bug fixes.

leenderthayen, 2021-03-02:
I haven't encountered any with the new version as you suggested, so I'm closing this.

closed 2021-03-02 by leenderthayen