JuliaPhysics/SolidStateDetectors.jl #136: Update config file format

fhagemann, 2021-02-12:
Based on the clarification of impurity and charge densities (#65), the redefinition of primitives (#84), the implementation of transformations (#62) and the refactoring of the CSG (#110), we need to update the config file format.

For this, we should decide on a long-term supported format, such that the user will have to update the config file format once (ideally with the help of conversion functions that SSD will provide).

lmh91, 2021-02-12:
At the end, so after the changes you mentioned, we should also have a look regarding the simulation of multiple detectors. 
This not only concerns the config file but maybe also `mutable struct Simulation`. 
Just to keep this in mind.

fhagemann, 2021-04-28:
Also, parameters of the charge drift model should be part of the general config file (preferably with splitted config files), rather than being stored separately in a folder that no one will find.

fhagemann, 2021-05-06:
I gathered all comments in PR regarding the new structure of config files to gather them here:

### PR #122 in a [comment from @lmh91](https://github.com/JuliaPhysics/SolidStateDetectors.jl/pull/122#issuecomment-739755931) about the general definition of `VolumePrimitives`
Instead of 
```yaml
type: "tube"
r: 
  from: 0.1
  to: 0.5
phi: 
  from: 0.0
  to: 360.0
h: 1.0
```
define the whole thing as `tube` instead of having a `type: "tube"`` inside the definition:
```yaml
tube:
  r: 
   from: 0.1
   to: 0.5
  phi:
   from: 0.0
   to: 360.0
  h: 1.0
```

### PR #84 in a [comment from @fhagemann](https://github.com/JuliaPhysics/SolidStateDetectors.jl/issues/84#issuecomment-758534887) about the general definition of `CSGTransformations`

Instead of making transformations part of the objects, i.e.
```yaml
type: "tube"
r: 
  from: 0.1
  to: 0.5
h: 1.0
translate:
  z: 1.0
```
give the object the name of the transformation and pass it the values needed for its definition, i.e.
```yaml
translate:
  z: 1.0  # defines the translation
  tube:   # primitive that is translated
    r: 
      from: 0.1
      to: 0.5
    h: 1.0
```
### PR #84 in a [comment from @lmh91](https://github.com/JuliaPhysics/SolidStateDetectors.jl/issues/84#issuecomment-758569474) on the general definition of `CSGRotations`

Rotations should be either defined by angles around specific axes or as a general rotation matrix:
```yaml
rotate: XYZ = [90, 0, 45] # first rotate around X, than, Y, than Z
rotate: ZX = [-45, 10] # first rotate around Z, than X
rotate: M = [1, 0, 0, 0, 0, 1, 0, 1, 0] # Vector of length 9 (if matrix notation is supported by yaml/json)
``` 
This requires the respective parsing functions for each of the cases.

### PR #84 in a [comment from @fhagemann](https://github.com/JuliaPhysics/SolidStateDetectors.jl/issues/84#issuecomment-758534887) about updating config files

The new version v0.6 should be equipped with two methods: one to convert **old** config files to the new format, and one to convert geometries created in code from the primitive constructors to a respective config file. First work can be found in the comment mentioned in the title of this subsection.

### PR #104 and #144 on splitted config files and `ChargeDriftModels`

The config files for detector geometries and `ADLChargeDriftModels` are spread all over the package and should be merged. 
Now with splitted config files, this would keep the config files short, clear and well structured. It would be nice to merge all config files together to avoid double defintions (material and temperature) and keep geometric input (like `phi110` for `ADLChargeDriftModel`)  next to the definition of the detector itself.
With this comes the idea to parse an arbitrary crystal orientation to `ADLChargeDriftModel` (#67) that should follow the same syntax as rotations in general.

fhagemann, 2021-05-15:
Before closing this issue, we still have to decide on:
- How (and where) to include `ChargeDriftModel` in the general detector config files (preferably using splitted config files).
- How to choose the config format such that multiple detectors can be defined. This also comes with a change in the source code of the `Simulation` struct.

fhagemann, 2021-07-07:
As #110 implies that we need to have a final discussion about the config file format, this is the issue to clarify!

lmh91, 2021-07-08:
As we discussed offline we see the format as fixed. 
We will add some features to it but those should not break old config files.

closed 2021-07-08 by lmh91

fhagemann, 2021-07-08:
Alright, then I will close this issue again.

fhagemann, 2021-07-11:
@lmh91: What is the default config file we would like the user to have?
- `json` or `ỳaml`?
- `surroundings` as part of the detector or outside?
- `origin` / `translate` ?

Then, I can update the script to convert old config files to new config files and force the new config files to have the format that we would recommend.

lmh91, 2021-07-12:
I would use `yaml`. It can be written more compact and one can add comments. 

I think we agreed that both is possible. I would put it outside in the example files. 

Even though I think `origin` is much better inside the primitives defintion you can use `translate` there.

PR #155 (2021-05-13): Update config file format
This pull request will fix #136 by updating the config file format.

### Remove `type` keyword from primitive dictionaries (9dc9abe)

For example, a `Tube` used to be defined like this:
```yaml
type: "tube"
r:
  from: 1.0
  to: 2.0
phi:
  from: 0.0
  to: 360.0
h: 2.0
```
and is now this:
```yaml
tube:
  r:
    from: 1.0
    to: 2.0
  phi:
    from: 0.0
    to: 360.0
  h: 2.0
```

### Remove `parts` keyword from `AbstractConstructiveGeometry` (9dc9abe)

For example, a `CSGUnion` used to be defined like this:
```yaml
type: "union"
  parts:
    - type: "tube"
      r:
   [... cut]