JuliaPhysics/SolidStateDetectors.jl #183: Proper handling of units in the user interface

fhagemann, 2021-07-15:
It would be nice to allow for units in the user interface, especially for plotting, e.g.
```julia
plot(simulation.electric_field, xlims = (-4u"cm", 4u"cm"))
```
This also comes with nice unit handling in configuration files (see #182).

oschulz, 2021-07-15:
We should take a look at [UnitfulRecipes.jl](https://github.com/jw3126/UnitfulRecipes.jl).

fhagemann, 2021-07-16:
I just realized this is a duplicate of #66. Closing #66 in favour of this.

lmh91, 2021-07-23:
Regarding the input and output units of the signal generation.

I think we should return the signal in units of (induced) charge and not "energy" anymore.
The conversion from charge to energy is done by the electronics which is not part of SSD.
The conversion from the energy of the hit table (input) to number of charge carriers 
should happen before the drift is simulated as the self-repulsion needs charge.

oschulz, 2021-07-23:
Sounds good.

fhagemann, 2021-10-05:
Adding [UnitfulRecipes.jl](https://github.com/jw3126/UnitfulRecipes.jl) as additional dependency fixes the plotting issue
(we can then just pass values with units to plot and they will automatically be added to the axes labels, quite neat!)

```julia
using Plots, Unitful, UnitfulRecipes
plot([1,2]u"m", [1,2]u"m", xlabel = "x", ylabel = "y", legend = false)
```
[image]
If you add more stuff with other units, it will keep the units from the first `plot` command
```julia
plot!([1200,1300]u"mm", [1000,1100]u"mm")
```
[image]
So, we can just add the internal units to all the objects/values when we plot them.

The question is how we want to deal with external units... Should we allow for a keyword argument in the plot recipe, something like `plot(..., length_unit = u"mm")`?

claudiaalvgar, 2025-10-28:
> [@claudiaalvgar](https://github.com/claudiaalvgar) could you take care of this?

Sure, I'll have a look at it

fhagemann, 2025-10-30:
Or, for the initial 0.11.0 release, we could restrict the new implementation of `CartesiaPoint{T}` and `CartesianVector{T}` to `T <: SSDFloat` or `T <: Real` instead of `T <: RealQuantity` or `T <: Number`, and then add unit support in a later 0.11.x release (?)