JuliaPhysics/SolidStateDetectors.jl #186: Painting contacts fails for specific configurations

fhagemann, 2021-07-22:
When updating the configuration files to avoid deprecation warnings, I stumbled across a possible bug (?) in the painting algorithm.

### Minimum working example

Configuration file: [`public_CGD_config_new.txt`](https://github.com/JuliaPhysics/SolidStateDetectors.jl/files/6859422/public_CGD_config_new.txt) --> rename to `public_CGD_config_new.yaml`
(Sorry, but files that end with `.yaml` cannot be uploaded easily here)
```julia
using SolidStateDetectors
T = Float32

sim = Simulation{T}("public_CGD_config_new.yaml")
simulate!(sim, convergence_limit = 1e-6, refinement_limits = [0.2], verbose = false)
evt = Event([CartesianPoint{T}(0,2e-3,0)])
simulate!(evt, sim, Δt = 1e-9, max_nsteps = 10000)
signalsum = sum(broadcast(wf -> abs(wf.value[end]), evt.waveforms))

@info signalsum
@assert isapprox( signalsum, T(2), atol = 1e-3 )
```
It looks like the second contact does not get painted onto the grid.

### Trouble shooting

The problem seems to be this `geometry` in the `difference` of the `contact` with `id: 2`:
```yaml
- box:
    widths: [0.5, 10, 10]
    origin: 
      x: 4.75
```
This does not happen when `widths: [0.6, 10, 10]`, when `òrigin: 4.8`, or when `T = Float64`. 
So, I am assuming that this is connected to some kind of rounding error (?)

The `geometry` in the initial configuration file `public_CGD_config.yaml` also works perfectly and looks like this:
```yaml
- box:
    x: 
      from: 4.5
      to: 5
    y: 10
    z: 10
```

lmh91, 2021-07-22:
I don't think the issue is the paining. At least not alone. 
Some debugging information I found out so far:

I disabled the painting and compared your config file with the example config file:

```julia
T = Float32
sim1 = Simulation{T}("public_CGD_config_new.yaml")
sim2 = Simulation{T}(SSD_examples[:CGD])
p_det = plot(sim1.detector);

apply_initial_state!(sim1, ElectricPotential, Grid(sim, max_tick_distance = 1u"mm"))
apply_initial_state!(sim2, ElectricPotential, Grid(sim, max_tick_distance = 1u"mm"))

plot(
    plot(sim1.point_types, y = 0, size = (600,600)),
    plot(sim2.point_types, y = 0, size = (600,600)),
    size = (1000, 400)
)
```
[image]

I compared the geometries of the first contact:
```julia
    @test sim1.detector.contacts[1].geometry.a.hX ==
          sim2.detector.contacts[1].geometry.a.hX
    @test sim1.detector.contacts[1].geometry.a.hY ==
          sim2.detector.contacts[1].geometry.a.hY 
    @test sim1.detector.contacts[1].geometry.a.hZ ==
          sim2.detector.contacts[1].geometry.a.hZ
    @test sim1.detector.contacts[1].geometry.a.origin ==
          sim2.detector.contacts[1].geometry.a.origin
    @test sim1.detector.contacts[1].geometry.a.rotation ==
          sim2.detector.contacts[1].geometry.a.rotation

    @test sim1.point_types.grid.x.ticks == 
          sim2.point_types.grid.x.ticks 
    @test sim1.point_types.grid.y == 
          sim2.point_types.grid.y 
    @test sim1.point_types.grid.z == 
          sim2.point_types.grid.z 
```

and only `hX` differs: 
`0.00025f0` vs `0.00025000004f0` 

The grid is the same though. Because the ticks are always rounded I believe.

lmh91, 2021-07-22:
So, paining should actually get rid of these numerical issues. 
I think I found the reason. A bug indeed...

closed 2021-07-23 by lmh91