JuliaPhysics/SolidStateDetectors.jl #280: Determination of the Electric Field at contacts

lmh91, 2022-04-08:
The issue came up from #279 

Right now, we assign an electric field vector to each grid point of the grid of the electric potential. 
The components of the vector is calculated by the differences in potential to the neighboring grid points in the respective dimensions.
However, for 2D-contacts, e.g. a plane in `x-y`, a wrong value is determined for the component in `z`: `E_z`.

#279 is a hotfix for this issue such that the value which corresponds to the side lying in the semiconductor is assigned to the grid point.
Which is okay to calculate the drift for charge carriers inside the semiconductor. 

However, in general this is not correct. 

I think instead of assigning electric field vectors to all grid points of the electric potential,
we should assign electric field vectors to each volume between the grid points of the electric potential.
Thus, the grid of the electric field would be smaller by one in each dimension 
as its ticks would be the midpoints of the ticks of the grid of the electric potential. 

This would also require changes in the interpolation of the electric field vector during the drift.

lmh91, 2022-04-08:
For the electric field energy calculation we already determine the electric field for each voxel
by the 8 grid points of the corners of each voxel:

https://github.com/JuliaPhysics/SolidStateDetectors.jl/blob/3cb4d4731f769f536a7befa0ba3d3488b5de1014/src/Simulation/ElectricFieldEnergy.jl#L22-L68

lmh91, 2022-04-08:
Alternatively, we could drop support (prohibit) 2D-objects. 

Which I am actually in favor of as our geometry is based on extended objects.

schustermartin, 2022-04-08:
> Alternatively, we could drop support (prohibit) 2D-objects.
> 
> Which I am actually in favor of as our geometry is based on extended objects.
> 
> Though, for certain (rotated) objects there still could be the case where the neighbors in one dimension of a contact grid point are both non-contact grid points, which, again, results in the 2D issue. E.g., at a corner of a cubic geometry of a rotated contact.

Yes, exactly, also as the contacts are painted onto a grid, situations may occur where only grid point in a dimension is assigned as contact. Just prohibiting 2D volumes does not necessarily solve this.

schustermartin, 2022-04-08:
> I think instead of assigning electric field vectors to all grid points of the electric potential, we should assign electric field vectors to each volume between the grid points of the electric [...]

I would not get rid of interpolation per se.
We could still interpolate, just not in voxels touching contact points. There we assign the field vector of the voxel to all points belonging to that voxel. Everywhere else interpolation can stay, and makes things smoother, especially in areas where the grid is coarse.

lmh91, 2022-04-08:
I just looked up the Interpolations.jl doc again.
They also offer a gradient method which might be the best way to go. 
```julia
using SolidStateDetectors, Unitful, Interpolations
sim = Simulation(SSD_examples[:Coax])
calculate_electric_potential!(sim)
itp = interpolate(broadcast(i -> sim.electric_potential.grid[i].ticks, (1,2,3)), sim.electric_potential.data, Gridded(Linear()));
pt = CylindricalPoint{Float32}(0.02, 0.01, 0.03)
Interpolations.gradient(itp, pt...)
```
Maybe we could also switch to other kind of interpolations than `Linear`. 
At least, like @schustermartin suggested for drift steps which are not close to an contact.

We interpolate anyhow already in the drift. But we could switch to interpolating the electric field vector directly from the electric potential by this.