JuliaPhysics/SolidStateDetectors.jl #316: Bug in normal(obj::(ConeMantle...)

bazzadino, 2022-08-25:
The transformation of "obj_normal"  back to the global coordinate system should be a pure rotation and not a  full _transform_into_global_coordinate_system(...)  (i.e.  rotation&translation ) that can stretch and change the length of obj_normal.
Removing the spurious translation has fixed some charge-drift problems I had in simulating a coaxial detector with the passivation on a cylindrical surface.
I suppose that the same argument applies also to ElipsoidMantle and ThorusMantle, but, sorry, I have not checked it in details.

fhagemann, 2022-08-25:
Thanks for reporting this bug. It seems like you have also modified the code locally to fix this bug yourself.
Can you have a look at the PR #317 and maybe test your code with this version of the code to see if that fixes the bug?

fhagemann, 2022-08-25:
PR #317 was just merged, I will trigger a new minor release v0.8.2 this evening such that the bugfix should be released latest tomorrow. Thanks for reporting and let us know if you find any more bugs or have any other questions regarding SolidStateDetectors.jl :)

PR #317 (2022-08-25): Fix bug when transforming `CartesianVector` to object coordinate systems
Should fix #316.

As stated correctly by @bazzadino, `CartesianVectors` should not be translated when being transformed to the global or object coordinate system. I have introduced new methods for these functions for `CartesianVector` that do not translate but only rotate the vectors and have modified the source code accordingly.