JuliaPhysics/SolidStateDetectors.jl #324: Estimate depletion voltage gives unreasonable value with wrong sign for segmented detectors

fhagemann, 2022-10-24:
If we add an impurity density to the example segmented `:BEGe` detector, the result of `estimate_depletion_voltage` is kind of odd..
```julia
using SolidStateDetectors
T = Float32
sim = Simulation{T}(SSD_examples[:BEGe])
sim.detector = SolidStateDetector(sim.detector, SolidStateDetectors.ConstantImpurityDensity{T}(1e16)) # 1e10 cm^-3
calculate_electric_potential!(sim, depletion_handling = true, refinement_limits = [0.2,0.1,0.05,0.03,0.01])
estimate_depletion_voltage(sim)
```
```
-537625.5f0 V
```

I would expect something positive of the order of `+2000V` here (?)...

lmh91, 2022-10-24:
Mind the keyword `bias_voltage_contact_id`:
`estimate_depletion_voltage(sim, bias_voltage_contact_id = 2)`.

But this is only implemented for 2 channel detectors.

Maybe a keyword `flip_bias_voltage_sign` could be added to solve this. 
(Just a quick thought)

fhagemann, 2022-10-24:
I am aware of the keyword, but the `bias_voltage_contact_id` is determined automatically as the one contact where a non-zero potential is applied. This should be the core contact of the segmented `:BEGe` detector here.

I guess the problem here is something else and I will try to explain them using some conceptional plots.

For a true coaxial detector, if you apply no potential to the contacts, the electric potential will peak somewhere in the detector due to the contribution from ionized impurities. In this example case, the maximum potential is reached somewhere in the center of the detector, resulting in electrons being attracted to neither of the contacts and the detector not being able to operate:
[image]

The idea of the algorithm to determine the depletion voltage is to apply a potential such that the local maximum is shifted towards one of the contacts:
[image]
In this example case, this happens at a potential of -21V applied to the contact on the right.

However, you can also (theoretically in this calculation) apply a positive potential to the contact on the right and also achieve that the maximum of the electric potential moves towards one of the contacts:
[image]
In this example case, this happens at a potential of +57V applied to the contact on the right.

It looks to me like the current algorithm tries out both of the things at the same time and returns the value with the higher absolute value (so here +57V). I wouldn't be surprised if you could theoretically deplete the example segmented `:BEGe` detector with -537625V, but I guess there is a second solution at around +2000V that also does the thing.

It might make sense to only return the solution with the same sign as the one initially applied to the contact, then?

fhagemann, 2022-10-24:
https://github.com/JuliaPhysics/SolidStateDetectors.jl/blob/7d79ad06eaf1b0648fcc208b0917e9d90d316304/src/Simulation/DepletionVoltage.jl#L131-L134

It looks to me like the current algorithm tries out both of the things at the same time and returns the value with the higher absolute value (so here +57V). I wouldn't be surprised if you could theoretically deplete the example segmented :BEGe detector with -537625V, but I guess there is a second solution at around +2000V that also does the thing.

It might make sense to only return the solution with the same sign as the one initially applied to the contact, then?

lmh91, 2022-10-24:
I just looked into it and it seems to be some numerical issue at the contacts.
Some values become `Inf`...

With the inverted coax it seems to work if you switch the impurity density I think.

fhagemann, 2022-10-24:
Any quick thoughts on how to resolve this?

lmh91, 2022-10-24:
I am not quite sure..

If I use the 5th contact as the contact where the bias voltage is applied (modify config file) I get reasonable values.

I think the reason might be because we divide through the values of the weighting potential.
https://github.com/JuliaPhysics/SolidStateDetectors.jl/blob/7d79ad06eaf1b0648fcc208b0917e9d90d316304/src/Simulation/DepletionVoltage.jl#L149
For contact 1 it is almost 0 a bit away from the point contact.
This seems to be unstable resulting in too high values.

If I also apply the HV to the point contact of the IVC the method also fails...

closed 2023-06-12 by fhagemann