JuliaPhysics/SolidStateDetectors.jl #397: G4EmStandardPhysics_option3/4 and Integration of the electric field in Geant4

hervasa2, 2024-08-08:
Full integration of SSD and geant4 could include the use of the electric field calculated by SSD. Particle propagation in Geant4 can then occur in the presence of this field. 
Field dependent spectrum effects (i.e. shift of single escape peak) could thus be simulated.
This would also require switching to `G4EmStandardPhysics_option4(or 3)` 
https://github.com/JuliaPhysics/SolidStateDetectors.jl/blob/30584dc6abab01085e47bcb21ad6672a7fe3896c/ext/Geant4/g4jl_application.jl#L5 
which should then also include doppler broadening effects (not captured by StandardPhysics). We would need some speed tests for this change.

I see two possible ways of passing the field to Geant4.

1. Following the guidance in [this thread](https://geant4-forum.web.cern.ch/t/creating-non-uniform-electric-field/2786/27) a non-uniform electric field can be created in Geant4 in the following manner.
```c
fFieldMessenger(nullptr)
{
// fEMfield = new G4UniformElectricField(G4ThreeVector(0.0,-1.0kilovolt/cm,0.0));
G4double zOffset = 0.0 * mm;
fEMfield = new PurgMagTabulatedEField3D(“PurgMag3D.TABLE”, zOffset);
fEquation = new G4EqMagElectricField(fEMfield);

fFieldManager = GetGlobalFieldManager();

UpdateIntegrator();
fFieldMessenger = new F02FieldMessenger(this);
}
``` 

- `PurgMag3D.TABLE` is just a text file with a field vector in each row: `x y z Ex Ey Ez`
- A `/field/getField` function needs to be defined to interpolate at any point, this can just be ported from the current methods in SSD

2. Create virtual volumes (given by grid size) of uniform electric field and use 
```c
fEMfield = new G4UniformElectricField(G4ThreeVector(0.0,100000.0*kilovolt/cm,0.0));
fEquation = new G4EqMagElectricField(fEMfield)
```

hervasa2, 2024-08-08:
Looks like `G4EmStandardPhysics_option4` works well and only causes a modest slowdown ~ 1.5-2 times slower 

```julia
window = 0.002
bw = 0.00025
plot(stephist(sum.(events.edep), bins = 1593.002-window:bw:1593.002+window),
    stephist(sum.(events.edep), bins = 2104.001-window:bw:2104.001+window),
    size = (1500,500)
    )
```
Double and single escape peaks with with `G4EmStandardPhysics_option4`
[image]

Double and single escape peaks with with `G4EmStandardPhysics`
[image]

oschulz, 2024-08-09:
Having doppler broadening in would be very useful when simulating calibration spectra, for example.

oschulz, 2024-08-09:
As for the electrical field effects on particle propagation, can you estimate if this will result in non-negligible corrections?

oschulz, 2024-08-09:
We should also be able it easy for the user to select the physics model with an option, right?

hervasa2, 2024-08-09:
> As for the electrical field effects on particle propagation, can you estimate if this will result in non-negligible corrections?

Yes I was wondering about this. We know that the shift of the single escape peak is about 0.1keV [(source)](https://doi.org/10.1016/0029-554X(72)90768-9) which is definitely observable. However I think we can ignore it for practical purposes. As for any other effects (i.e. shift of positron annihilation position due to drift, charged particle tracks...) its hard to say a priori.

oschulz, 2024-08-09:
Let's split this - user selectable physics lists should be easy to do, while incorporating the electrical field will be more involved (are you interested in pursuing this @hervasa2 ?).

hervasa2, 2024-08-09:
Yes, I will work on selectable physics lists next week! See you at volleyball!

oschulz, 2024-08-09:
@hervasa2 don't worry about the physics lists, @fhagemann will do this over the weekend or so, it's quick thing.

Incorporating the electrical field seems like a different best though, someone will need to contribute this.

tdixon97, 2025-04-15:
This is interesting and something we thought about for remage, did anyone look into this more?

fhagemann, 2025-04-15:
@benedikt-nagler is looking into this, but no real results yet.

closed 2026-08-27 by fhagemann