JuliaPhysics/SolidStateDetectors.jl #420: Implementation of new material for semiconductor

Padniuk, 2024-10-11:
As I understood for now there are only two materials which can be used for semiconductor: HPGe and Si. In other way we will see the follow:
```
ERROR: LoadError: "Material characteristics for \"$(material)\" not defined yet"  
```
I want to use another materials such as CZT(Cadmium Zinc Telluride) or even more custom material. Of course we need to specify other properties of such materials: charged carriers creation energy, Fano factor etc.

I know that such [mechanism](https://allpix-squared.docs.cern.ch/docs/13_faq/05_development/#how-do-i-add-a-new-sensor-material) has been already implemented in Allpix Squared(framework for CMOS).  Shortly, the idea is to create and register material as you do it in Geant4. Then add material properties and that's it.
 
I guess that I  should do something similar, but I'm  not a huge expert in Julia. 

I would appreciate any suggestions to implement such functionality. Thank you very much!

oschulz, 2024-10-12:
Hi @Padniuk , using "CdZnTe" should work. We do have it in

https://github.com/JuliaPhysics/SolidStateDetectors.jl/blob/main/src/MaterialProperties/MaterialProperties.jl

though I think it's not been carefully validated yet. But if you know CdZnTe and could over the values, your input would be very welcome!

Padniuk, 2024-10-12:
Thank you for your quick answer. I will look for the precise values. Now properties definition is clear.
 
One more interesting thing is the custom semiconductors. As I see you [parse](https://github.com/JuliaPhysics/SolidStateDetectors.jl/tree/8bbf0e831154c863e668c252f049d8ddd1f525e5/ext/Geant4/io_gdml.jl#L455-L468) materials from Geant4. But what if I need define custom semiconductor which is not implemented in Geant4. We know that basic Geant contains Si, Ge, CZT, GaAs itself and we can simply load them by parsing. But imagine situation when I have custom semiconductor: I know the chemical composition and properties. We know about implementation of properties for such material. Before that I need to create material, such we do it in C++:
```cpp
  G4Material* ABC = new G4Material("ABC", density, 3);
  ABC->AddElement(A, 2);
  ABC->AddElement(B, 1);
  ABC->AddElement(C, 4)
```
Then I should register this material as you do it in parsing, and hopefully that's it. As I mentioned you parse materials from Geant4.jl which is wrapper for Geant4(standard, C++). I guess that probably I need to define material in standard Geant4 and rebuild all packages. 

I do not think that it is a good idea, cause I have installed standard Geant4 v10.6, julia uses another one:
```julia
julia> using Geant4

julia> runManager = G4RunManager()
**************************************************************
Geant4 version Name: geant4-11-02-patch-01 [MT]   (16-February-2024)
Copyright : Geant4 Collaboration
References : NIM A 506 (2003), 250-303
: IEEE-TNS 53 (2006), 270-278
: NIM A 835 (2016), 186-225
WWW : http://geant4.org/
*************************************************************
```
So, I need to search where is  **geant4-11-02-patch-01** and change it. I guess it should be much easier way to do it.

Sorry, for such naive questions, I just want to do all properly

oschulz, 2024-10-12:
There's no need to recompile Geant4 to add materials in general. SolidStateDetectors.jl provides material definitions to Geant4 via GDML. @fhagemann can we extend the SSD config files to allow for adding additional info that Geant4 needs to custom semiconductor/material definitions?

If you use Geant4.jl directly, there should also be no need to recompile Geant4 to add materials, Geant4 allows for adding materials at runtime. I'm not sure the Geant4 Julia wrapper allows this already, but if not, that should be a minor addition.

Padniuk, 2024-10-12:
@oschulz, first of all I want to thank you for your answers. I implemented test material with test properties => code is run.

I want to mention small bug. Initially, before our discussion I tried to run code with the CZT as semiconductor and I have got an error:
```
ERROR: LoadError: "Material characteristics for \"Cadmium zinc telluride\" not defined yet"
``` 
After that I thought that probably I do not understand something important. Our discussion showed me that the reason of my doubts was the following bug: [name in material_properties for CZT](https://github.com/JuliaPhysics/SolidStateDetectors.jl/blob/8bbf0e831154c863e668c252f049d8ddd1f525e5/src/MaterialProperties/MaterialProperties.jl#L86) does not match to the [condition name in parse method](https://github.com/JuliaPhysics/SolidStateDetectors.jl/blob/8bbf0e831154c863e668c252f049d8ddd1f525e5/ext/Geant4/io_gdml.jl#L465). After changing it to the "CdZnTe" error is lost. I made this conclusion by comparison between the [name in material_properties for HPGe](https://github.com/JuliaPhysics/SolidStateDetectors.jl/blob/8bbf0e831154c863e668c252f049d8ddd1f525e5/src/MaterialProperties/MaterialProperties.jl#L30) and [condition name in parse method](https://github.com/JuliaPhysics/SolidStateDetectors.jl/blob/8bbf0e831154c863e668c252f049d8ddd1f525e5/ext/Geant4/io_gdml.jl#L457). Before that I thought [this dictionary corresponds for such things](https://github.com/JuliaPhysics/SolidStateDetectors.jl/blob/8bbf0e831154c863e668c252f049d8ddd1f525e5/src/MaterialProperties/MaterialProperties.jl#L95) but now I realize that it just allows to use specific strings instead of exact key and does not have any impact to the dict value attributes such as name.

Probably these will be the last questions from this topic(Sorry that I disturb you a lot):
- I saw that for HPGe(and Si) you use [AbstractDriftMaterial](https://github.com/JuliaPhysics/SolidStateDetectors.jl/blob/8bbf0e831154c863e668c252f049d8ddd1f525e5/src/MaterialProperties/MaterialProperties.jl#L23-L24) abstruct type. Is it implemented just to specify that [ADLChargeDriftModel](https://github.com/JuliaPhysics/SolidStateDetectors.jl/tree/8bbf0e831154c863e668c252f049d8ddd1f525e5/src/ChargeDriftModels/ADLChargeDriftModel) can be used only for HPGe and Si? (e.g. In case of another semiconductor I should do the same manipulations with abstract type or, what looks more correct, I should implement own model)
- Previous question forces to ask about similar challenges which should be solved for correct implementation. As I understood from the [slide 3](https://indico.cern.ch/event/1410341/contributions/6135575/attachments/2938025/5160938/JuliaHEP%202024.pdf#page=4) there is only 2 things which connected to the material: charge drift model and impurities distribution. Other things depend on geometry and bias and independent on the material. Am I right?

oschulz, 2024-10-13:
> name in material_properties for CZT does not match to the condition name in parse method

Oops, yes, I think that's a bug. Geant4 support in SolidStateDetectors is still very new. Can you try changing that in io_gdml.jl and see if it works, then? Longer term we should add a more extensible mechanism.

> I saw that for HPGe(and Si) you use AbstractDriftMaterial. Is it implemented just to specify that ADLChargeDriftModel can be used only for HPGe and Si?

No, `CdZnTe` should indeed be an `AbstractDriftMaterial` as well, so an `AbstractChargeDriftModel` can be defined for it. Not every charge drift model will be appropriate for every semiconductor, of course, and  CdZnTe is quite different from HPGe and Si. So I guess the `ADLChargeDriftModel` may not be the best default for it ;-) ... do you know which drift model is state of the art for CdZnTe? It's easy to add more drift models to SolidStateDetectors, if the math is known.

We added CdZnTe because someone wanted to play around with it, and that was mainly for field calculations, if I remember correctly. I think you're probably the first serious user of CdZnTe in SolidStateDetectors, so some fixes may be required. :-)

Padniuk, 2024-10-15:
``` Can you try changing that in io_gdml.jl and see if it works, then?```
It works. Probably I did not explain correctly in the previous message that I have already done these changes and all is fine.

I dove deeper into the models which you are using for charge drift. And yes, it was too naive to hope that ```ADLChargeDriftModel``` will suit to my case. 

For now I used the modified ```ElectricFieldChargeDriftModel```. Default model uses mobility 10 000 $\frac{cm^2}{V\cdot s}$. I just added scale factor to [electron and hole velocity function](https://github.com/JuliaPhysics/SolidStateDetectors.jl/blob/main/src/ChargeDriftModels/ElectricFieldChargeDriftModel/ElectricFieldChargeDriftModel.jl#L13-L14) to get precise value of mobilities. For CZT it shows reasonable rise time, signal shape and amplitude comparing to the experiment. Of course, it does not mean that this approach is perfect, that's why now I am trying to find more precise way how to describe charge drift.

In any case I understood a lot during this discussion. Thank you for your answers, you are doing a big deal with this framework! I wish you all the best!

oschulz, 2024-10-15:
Thanks @Padniuk - would you like to make a pull request with your changes? Or just send the relevant code lines here, so we can add them?

Padniuk, 2024-10-17:
I made a pull request. I use Python and C++ often, and I am to a big expert in Julia. So the code styling can be not so good, but logically all works fine. I called this drift model as ``` IsotropicMobilityChargeDriftModel ```

closed 2024-11-05 by Padniuk