JuliaPhysics/SolidStateDetectors.jl #423: Implementation of bubble depletion check in the "estimate_depletion_voltage" function

zpdg, 2024-10-18:
Hi, everyone. I use SSD.jl (V0.10.2) to simulate a p-type BEGe detector ($r \times h= 40 \times 42.6$ mm, bias voltage = 4500 V). The expected depletion voltage would be above 3000 V.

However, the result of `estimate_depletion_voltage` is around 1400 V. So, I change the bias to 2000 V and the `is_depleted` does return a "true", but there is a bubble in the center of the detector (in the field plot). The hole might end up in the bubble instead of the contact. So, 2000 V might be able to deplete the detector but not enough to get rid of the bubble.

The code "estimate_depletion_voltage" only makes sure that the potential in the detector is within the range of 0 ~ bias voltage but couldn't avoid the bubble inside, while the bubble reflects a local minimum of potential inside detector (not on the boundaries). 

The config is uploaded in the siggen format. (I load it with `sim=Simulation{T}(siggentodict(readsiggen("./test.txt")))`)
[test.txt](https://github.com/user-attachments/files/17429236/test.txt)

I found that the issue #250 and #311 has talked about the problem. The docstring in the code of `old_estimate_depletion_voltage` shows the idea of the bubble check ("No bubble" should be equivalent to "the gradient of the superposition field $EP_i + U_D * EP_0$ becomes nowhere zero inside").  However, the result of `old_estimate_depletion_voltage` is also around 1400 V.
[image]

I also used the siggen to calculate the depletion voltage. The result is 3560 V, which seems correct. There are three steps to calculate the depletion voltage  in siggen:
1. find the voltage that makes sure "the gradient of the superposition field $EP_i + U_D * EP_0$ becomes nowhere zero inside" 
[image]
2. find the voltage that make sure no bubble exists
[image]
3. The depletion voltage is chosen to be the bigger one
 

Is my understanding of this correct? Will you add the implementation of the bubble depletion check in the "estimate_depletion_voltage" function? I can contribute on this, but it might take me some time since I am not very familiar to julia.

fhagemann, 2024-10-18:
Sometimes you get weird results when the potentials defined in the config files are below the depletion voltage.

When you calculate the electric potential before determining the depletion voltage, do you set depletion_handling=true ?

zpdg, 2024-10-18:
> Sometimes you get weird results when the potentials defined in the config files are below the depletion voltage.
> 
> When you calculate the electric potential before determining the depletion voltage, do you set depletion_handling=true ?

Yes, I have set depletion_handling=true. And I got the incorrect depletion voltage (1400 V) under a big enough bias voltage (4500 V). You can reproduce this with the config file I upload (change the bias voltage in the file to 4500 V first).

My understanding is that the function `estimate_depletion_voltage` doesn't consider the bubble effect, so the predicted depletion voltage is smaller than expected. Besides, the bubble effect is more likely to appear on the BEGe. I have tested with the public IVC config (public_ppc_config_SigGen.config), and the depletion voltage seems correct and in agreement with the result from mjd_siggen.

fhagemann, 2024-10-18:
Alright thanks, we will have a look into this

closed 2024-11-15 by fhagemann