JuliaPhysics/SolidStateDetectors.jl #454: How to plot sim.electric_field in 3d

rajxabc, 2025-02-07:
How can i plot sim.electric_field in 3D, and is it possible to set values above and below certain range transparent ?
or oblique slices (horizontal and vertical ) in 3d plot 
Thanks

fhagemann, 2025-02-07:
Not with the current plot recipes, but you can have a lot at either Plots.jl or Makie.jl, to see if they allow to plot this. You would then need to access the (probably unevenly spaced) grid points and the data using `sim.electric_field.grid` and `sim.electric_field.data`.

If you have an example plot that you want to reproduce, I also can look into this myself and see if I can provide something

rajxabc, 2025-02-10:
something like the attached files (oblique multiple cross-section ) or in 3d plot where min max value can be threshold 
[image]
[image]

simplest recipe , will be much appreciated
Thanks

fhagemann, 2025-02-10:
I just had a quick look at this, and can provide you something.
Feel free to modify it accordingly:

```julia
using SolidStateDetectors
using Plots; pyplot()
using LinearAlgebra
using Unitful

T = Float32
sim = Simulation{T}(SSD_examples[:CGD])
simulate!(sim)

clims = (0,1000)

# Cross-section in xz-plane at y = 0
y = 0
yidx = SolidStateDetectors.searchsortednearest(sim.electric_field.grid[2], T(y))
y = sim.electric_field.grid[2][yidx]

xs = sim.electric_field.grid[1].ticks
ys = sim.electric_field.grid[2].ticks
zidx = findall(z -> z >= -0.00275, sim.electric_field.grid[3].ticks)
zs = sim.electric_field.grid[3].ticks[zidx]

xx, yy, zz = xs' .* ones(length(zs)) * u"m",  (xs' .* zeros(length(zs)) .+ y) * u"m", zs .* ones(length(xs))' * u"m"
surface(xx, yy, zz, fill_z = clamp.(norm.(sim.electric_field.data[:,yidx,zidx])' ./ 1000, clims...), clims = clims)

# Cross-section in xy-plane at z = -0.00275
z = -0.00275
zidx = SolidStateDetectors.searchsortednearest(sim.electric_field.grid[3], T(z))
z = sim.electric_field.grid[3][zidx]

xs = sim.electric_field.grid[1].ticks
yidx = findall(y -> y <= 0, sim.electric_field.grid[2].ticks)
ys = sim.electric_field.grid[2].ticks[yidx]
zs = sim.electric_field.grid[3].ticks

xx, yy, zz = xs' .* ones(length(ys)) * u"m",  (ones(length(xs))' .* ys) * u"m", (xs' .* zeros(length(ys)) .+ z) * u"m"
surface!(xx, yy, zz, fill_z = clamp.(norm.(sim.electric_field.data[:,yidx,zidx])' ./ 1000, clims...), clims = clims)
```

[image]

The `surface` method somehow does not seem to work with `gr`, so I used `pyplot` here. You can also check for other backends.

fhagemann, 2025-02-10:
If this turns out to be useful, we can think about including a plot recipe for plotting cross-sections of the potentials and fields in the SolidStateDetectors package.

rajxabc, 2025-02-10:
Thanks for the right direction , this looks good.
I will modify according to my specific need .
Thanks again