JuliaPhysics/SolidStateDetectors.jl #457: Error parsing Torus in Geant4

hervasa2, 2025-02-12:
Events are recorded outside of the detector in the following simulation. From the energy depositions it is clear that a negative hollow torus is being placed incorrectly at the bottom outer edge the detector (and the negative cone ignored). The torii at the top outer edge (top rounding) and and point contact are also ignored.
```julia
detector_config_filename = "ppc.json"
T = Float32
sim = Simulation{T}(detector_config_filename);
source_1 = MonoenergeticSource(
    "gamma",                              # Type of particle beam
    2.615u"MeV",                          # Energy of particle
    CartesianPoint(0.1, 0., 0.02),      # Location of the source
    CartesianVector(-1,0,0),              # Direction of the source
    31u"°"                                # Opening angle of the source emission
)
plot(sim.detector, size = (200,200), guidefontsize = 1, tickfontsize = 1, axis = false, grid = false, 
    camera = (0,0), xlims = (-0.04,0.11), zlims = (-0.04,0.11))
plot!(source_1, ms = 2)
```

[image]

```julia
app = G4JLApplication(sim, source_1, verbose = false)
N_events = 1000000
events = run_geant4_simulation(app, N_events)

tol = 4u"mm"
y_slice = 0u"mm"
z_slice = 0u"mm"
binning = 0.2
idx1 = findall(abs.(getproperty.(events.pos.data, :y) .- y_slice) .< tol)
plot1 = histogram2d(getproperty.(events.pos.data[idx1], :x),
    getproperty.(events.pos.data[idx1], :z),
    aspect_ratio = 1,
    bins = (-40:binning:40, 0:binning:42))
plot!(sim.detector, st = :slice, y = y_slice, ms = 0.5)

idx2 = findall(abs.(getproperty.(events.pos.data, :z) .- z_slice) .< tol)
plot2 = histogram2d(getproperty.(events.pos.data[idx2], :x),
    getproperty.(events.pos.data[idx2], :y),
    aspect_ratio = 1,
    bins = (-40:binning:40, -40:binning:40))
plot!(sim.detector, st = :slice, z = z_slice, ms = 0.5)

plot(plot1, plot2)
```
[image]

Note that the point type map indicates the desired geometry of the detector.

[image]

 Config file is included here [ppc.json](https://github.com/user-attachments/files/18768865/ppc.json)

fhagemann, 2025-02-12:
Yes, the reason being that the current implementation cannot handle `θ` ranges that are not 0° to 360° yet:
https://github.com/JuliaPhysics/SolidStateDetectors.jl/blob/3997df116196f27551a25592b84b215c61a8e599/ext/Geant4/io_gdml.jl#L325-L353

Let me see whether this is easy to implement right now.

fhagemann, 2025-02-12:
So, the issue here seems to be that some volumes are flagged as having no volumes.
- The negative cone at the bottom outer edge is ignored because `rmin` and `rmax` had the same value at the top (this is a bug and should be fixed in #458). 
- The p+ contact torus was ignored because `r_torus == 0`.
  In general, I would recommend to use a `sphere` here, but I also relaxed the criteria in `has_volume(::Torus)` to allow for `r_torus == 0` in #458.
- All tori with `θ` ranges that are not 0° to 360° are parsed incorrectly. For now, I would suggest to use `intersection` or `difference` to deal with this and we will have a look to support partial-`θ` tori in the near future.
- The reason why the hollow torus was placed wrongly is because some volume primitives (bottom outer cone, p+ contact torus) were skipped because of being flagged as having no volume. However, their translations were still combined together. This is indeed a bug that will need some deeper look into.

I made slight modifications to your config file, changing the lines that triggered the errors mentioned above: [ppc.json](https://github.com/user-attachments/files/18771785/ppc.json)
This config file should work with the current version of SSD (without the fixes in #458).

This results in a reasonable plot (with `N_events = 20_000` and `binning = 0.5` to get the plot faster):
[image]

hervasa2, 2025-02-12:
Thank you Felix. And for the working config file!
I did some more digging.
I removed the torii and still found the problem with the negative cone. But this as you say is expected. Now however there is also a problem with the negative tube for the point contact. 

[image]

[ppc.json](https://github.com/user-attachments/files/18772091/ppc.json)

fhagemann, 2025-02-12:
Yes, this is a bug because the negative cone is flagged as (`has_volume == false`), and therefore, the origin of the negative cone is passed onto the contact tube. (so its moved `2.25mm` up instead of the negative cone).
I hope that @benedikt-nagler can find some time to look at this.

For now, you can run the code with the current `main` branch that has the fix of #458, or you can set the outer radius of the negative cone to something slightly bigger than `35.5`, e.g.
```json
"cone": {
    "r": {
        "bottom": {
            "from": 30.9,
            "to": 35.6
        },
        "top": {
            "from": 35.5,
            "to": 35.6 
        }
    },
    "h": 4.6,
    "origin": [
        0.0,
        0.0,
        2.25
    ]
}
```
Note that I increased all `to` values from `35.5` to `35.6`

fhagemann, 2025-02-12:
> Thank you Felix. And for the working config file! I did some more digging. I removed the torii and still found the problem with the negative cone. But this as you say is expected. Now however there is also a problem with the negative tube for the point contact. It seems to be shifted upward.
> <img alt="Image" width="897" src="https://private-user-images.githubusercontent.com/41748838/412550042-9b70399c-e6b2-46cf-9019-a852f67d43b3.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzkzODYwMTcsIm5iZiI6MTczOTM4NTcxNywicGF0aCI6Ii80MTc0ODgzOC80MTI1NTAwNDItOWI3MDM5OWMtZTZiMi00NmNmLTkwMTktYTg1MmY2N2Q0M2IzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAyMTIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMjEyVDE4NDE1N1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWIzMGMyZGU4MzQ3NmJkMGViOTZmYzlkNjZlYTNmZjQ4ODE0NTJiMjZjMzEyMGMyYmQzMGNhMjQzNzc3ZDVjZTImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.o9Cj7j7am5o8toNzkZRy1eFS6lj0dQy6YqpCwwtrzGk">

@hervasa2 This bug should be fixed by #459.

fhagemann, 2025-02-12:
Now the only thing that remains is to implement tori with partial `θ`-ranges in the Geant4 extension.

fhagemann, 2025-02-13:
So, after #459, everything looks fine to me
```julia
N_events = 100_000
events = run_geant4_simulation(app, N_events)
all(p -> CartesianPoint(ustrip.(u"m", p)) in sim.detector, SolidStateDetectors.flatview(events.pos))
```
```
true
```

hervasa2, 2025-02-13:
Looks perfect now. Thank you for looking at it so fast!
[image]

fhagemann, 2025-02-17:
Thanks for reporting! :)