JuliaPhysics/SolidStateDetectors.jl #499: Error Doing Multiple Geant4 Simulations in one Julia REPL Session

DineshGangavarapu, 2025-06-13:
I am working at ORNL doing Julia (1.11) Geant4 simulations for minimizing PEN enclosure thickness. I keep encountering a recurring error where I am unable to perform more than one Geant4 simulation in a single Julia session. If I were to run my code, which includes two sources and a simulation for each, line by line, it would break down when I perform the second Geant4 simulation with a `UndefRefError`. This has been recurring and I have not been able to find a solution to it. Please find attached below a minimal working example. 
``` julia
using Geant4
using SolidStateDetectors
using Unitful
source_1 = MonoenergeticSource(
    "e-",                                 
    3.5u"MeV",                            
    CartesianPoint(0.065, 0., 0.05),      
    CartesianVector(-1,0,0),               
    10u"°"    ) 
source_2 = MonoenergeticSource(
    "e-",                                 
    2.25u"MeV",                            
    CartesianPoint(0.085, 0., 0.15),      
    CartesianVector(-2,0,0),               
    15u"°"    ) 
T = Float32
sim = Simulation{T}(SSD_examples[:InvertedCoaxInCryostat])
app = G4JLApplication(sim, source_1, verbose = false)
events = run_geant4_simulation(app, 1000)
sim_2 = Simulation{T}(SSD_examples[:InvertedCoaxInCryostat])
app_2 = G4JLApplication(sim_2, source_2, verbose = false)
events_2 = run_geant4_simulation(app_2, 1000)
```

CC @theHenks

fhagemann, 2025-06-13:
This is a known issue.
You do rely on having both Geant4 simulations running in the same session? If not, i would suggest to just run two separate simulations for source 1 and source 2

fhagemann, 2025-06-13:
You can create maximum one G4JLApplication per Julia session, but you can definitely run multiple event generations with the same G4JLApplication within one session